afiskon/stm32-ssd1306 #2: Set memory addres mode

CoBa31Rus, 2018-12-09:
Hello, in code i see this:
`ssd1306_WriteCommand(0x20);//Set Memory Addressing Mode `
`ssd1306_WriteCommand(0x10); // 00,Horizontal Addressing Mode; 01,Vertical Addressing Mode;`
`// 10,Page Addressing Mode (RESET); 11,Invalid`
But datasheet say:
> A[1:0] = 01b, Vertical Addressing Mode

that mean bin, not hex.
RESET state is 0b10, that mean this code not need.
P.S. Sorry for my eanglish)

afiskon, 2019-01-22:
Hello @CoBa31Rus 

Thank you for reporting this! And I'm sorry it took me so long to write a reply.

You are probably right. I'm afraid at the moment I don't have spare time to check this properly on all supported devices.

Since the issue doesn't seem to be critical let's just keep it open for now.

afiskon, 2019-12-14:
The code was fixed in f8df7d2. Thanks for reporting!

commit f8df7d2d (2019-12-14): Fix #2 reported by @CoBa31Rus
closed 2019-12-14 by afiskon