afiskon/stm32-ssd1306 #3: Fonts

gc9n, 2019-01-04:
Hey  Guys ,
IS there any program to generate fonts ,  i want to import a custom font that i want, is this possible??

afiskon, 2019-01-05:
Hello,

The fonts were borrowed from another (MIT licensed) library https://github.com/4ilo/ssd1306-stm32HAL I don't know what software what used to create them. Personally I would probably write a little script in Python.

closed 2019-01-05 by afiskon

afiskon, 2022-11-04:
There is a font generator developed by @the-this-pointer https://github.com/the-this-pointer/glcd-font-calculator

At the moment of writing only Windows is supported.

afiskon, 2023-01-23:
A tutorial about creating custom fonts was added: https://github.com/afiskon/stm32-ssd1306/tree/master/examples/custom-fonts