afiskon/stm32-ssd1306 #5: SPI & STM32 CLK frequency

metalalive, 2019-02-11:
Hello
thanks for sharing your code, 
I am new and learning to initialize ssd1306 OLED module which can run at 8 MHz with SPI interface , 

When testing with SPI interface :
- Did you test the code on STM32 running at 16MHz (using HSI clock source) or some other frequency / other clock source ? 
- What's the clock frequency on your (SCL) ssd1306 OLED ?

because in this [line of code](https://github.com/afiskon/stm32-ssd1306/blob/02a7a05223893ca98743428c161ab08fb7318e24/examples/spi/stm32f4/Src/main.c#L152) I saw the baudrate scaler is set to 64.
I have not figured out how to calculate for correct SPI clock.

could you please give some tips / hints ?
thanks for your help

afiskon, 2019-02-11:
Hello @metalalive 

Thanks for the interest for this library. I tested it with various clock speeds and it worked fine.

The line of code you are mentioning was generated in STM32CubeMX. Here are all resulting clock speeds:

[image]

However the library doesn't depend on actual SPI clock speed.

closed 2019-02-11 by afiskon

metalalive, 2019-02-11:
thanks for your reply, I think I found the way to check SPI baudrate by [this thread](https://community.st.com/s/question/0D50X00009XkX8gSAF/calculate-the-baudrate-in-spi) , and the clock configuration in STM32CubeMX (as you mentioned) , and stm32 reference manual. :))