afiskon/stm32-ssd1306 #9: Adding support for nRF52

DuyTrandeLion, 2019-07-22:
Hello,

I am using your library on STM32 and I just finished porting it for my Nordic nRF52.
May I re-upload the library and link it to your library? 

Thank you and best regards,
      Duy

afiskon, 2019-07-22:
Hello @DuyTrandeLion 

Sure. This library has MIT license and you can do it. Please send a link to your library and I will add it to README as well.

closed 2019-07-22 by afiskon

DuyTrandeLion, 2020-02-16:
Hello @afiskon ,
Can you add my library for the nRF52 to your link? I added yours to my README.
https://github.com/DuyTrandeLion/nrf52-ssd1309
Thanks and best regards,
         Duy

afiskon, 2020-02-17:
@DuyTrandeLion Sure. README was updated. Nice work!

Could you please correct my name? :)

> developed by OAleksander Alekseev

DuyTrandeLion, 2020-02-17:
@afiskon Thank you. Already update your name in README.