afiskon/stm32-ssd1306 #12: Solution for 32row OLEDs

wmajst, 2019-09-08:
I've used your library for 128x32 OLED,
however it required following modifications in ssd1306_init():

    ssd1306_WriteCommand(0xA8); //--set multiplex ratio(1 to 64) - CHECK
#if SSD1306_HEIGHT==32
    ssd1306_WriteCommand(0x1F); //
#else
    ssd1306_WriteCommand(0x3F); //
#endif

    ssd1306_WriteCommand(0xDA); //--set com pins hardware configuration - CHECK
#if SSD1306_HEIGHT==32
    ssd1306_WriteCommand(0x02);
#else
    ssd1306_WriteCommand(0x12);
#endif

vakochetkov, 2019-12-26:
Ya, this little fix works with 128x32 OLED. Otherwise you have to invert WIDTH and HEIGHT values in header file.

afiskon, 2020-03-18:
I believe it was solved in #22.

closed 2020-03-18 by afiskon