afiskon/stm32-ssd1306 #14: PlatformIO can't override / define macros

scottc11, 2019-11-13:
I'm looking to make this library compatible with [platformIO](https://platformio.org) and for the most part I have the library being recognized but I can't seem to override the macros

```
// src/main.h
#define SSD1306_USE_I2C
#define SSD1306_I2C_PORT                       hi2c1
#define SSD1306_I2C_ADDR                       (0x3C << 1)
#define SSD1306_HEIGHT                         32
#define SSD1306_WIDTH                          128

// src/main.c
#include "ssd1306.h"
```

These are the errors I am getting when I compile.

```
In file included from lib/SSD1306/ssd1306/ssd1306.c:1:0:
lib/SSD1306/ssd1306/ssd1306.h:73:2: error: #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro!"
 #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro!"
  ^~~~~
lib/SSD1306/ssd1306/ssd1306.c:49:2: error: #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro"
 #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro"
  ^~~~~
*** [.pio/build/bluepill_f103c8_128k/libec4/SSD1306/ssd1306/ssd1306.o] Error 1
Compiling .pio/build/bluepill_f103c8_128k/FrameworkHALDriver/Src/stm32f1xx_hal.o
In file included from lib/SSD1306/ssd1306/ssd1306_tests.c:1:0:
lib/SSD1306/ssd1306/ssd1306.h:73:2: error: #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro!"
 #error "You should define SSD1306_USE_SPI or SSD1306_USE_I2C macro!"
  ^~~~~
*** [.pio/build/bluepill_f103c8_128k/libec4/SSD1306/ssd1306/ssd1306_tests.o] Error 1
```

I'm going to get my own version working, but if someone would like to help me with this I can make a pull request so the platformIO community can start using it 👍

afiskon, 2019-11-14:
Hello @scottc11 

Thanks for the effort and  the interest for the library!

These macros supposed to be defined using compiler arguments, like this https://github.com/afiskon/stm32-ssd1306/blob/master/examples/spi/stm32f1/Makefile#L127 Unfortunately I'm not familiar with PlatformIO but I'm sure there are corresponding project settings somewhere in UI.

SwapUNaph, 2019-12-24:
@scottc11 
Just add the following lines to `ssd1306.h` :
`
#define STM32F1
#define SSD1306_USE_I2C
`

@afiskon I liked the library very much. I would like to extend it to do some more graphic stuff like drawing shapes and animation. What do say?

afiskon, 2019-12-24:
@SwapUNaph patches are welcome, but please don't break the backward compatibility.

ChaitanyaGhate, 2020-12-11:
Hello folks,
I want to use this library for SSD1309 OLED than what changes I have to do?
I am using STM32F401RC.

afiskon, 2021-01-16:
@scottc11 I'm closing this discussion since @SwapUNaph proposed a solution. Please let us know if it didn't work for you.

@ChaitanyaGhate the library works with SSD1309 as is. Please read README carefully and follow the instructions. Let us know if you have a particular issue with a particular LCD. Also if you believe that README can be improved, patches are welcome!

closed 2021-01-16 by afiskon