afiskon/stm32-ssd1306 #26: Support for more display resolutions

disposedtrolley, 2020-03-28:
Hello, thanks for taking the time to put together this library!

I'm trying to use this library with this [1.12" OLED breakout from Pimoroni](https://shop.pimoroni.com/products/1-12-oled-breakout?variant=12628508704851)
which uses the `SH1106` controller. The display has a resolution
of 128x128 which exceeds the maximum height supported by this
library. This is the error displayed when I attempt to set the
`SSD1306_HEIGHT` to 128:

    `error: #error "Only 32 or 64 lines of height are supported!"`

Are there plans to support additional display resolutions? I'm
also more than happy to try and contribute the feature - although
I'm very new to embedded development so I'll need some pointers!

afiskon, 2020-03-31:
Hi @disposedtrolley ,

Indeed, SH1107-based displays are not supported at the moment. Your contribution would be most appreciated. The datasheet ( https://www.displayfuture.com/Display/datasheet/controller/SH1107.pdf ) seems to be written well and apparently it has all the information you might need. Basically you only need to replace the "32/64 lines" check to "32/64/128 lines" check and maybe change a few commands. Good luck!

afiskon, 2020-03-31:
Closed by https://github.com/afiskon/stm32-ssd1306/pull/27

closed 2020-03-31 by afiskon