afiskon/stm32-ssd1306 #36: SH1106 compatibility

Dreamy-Z3r0, 2021-01-13:
I notice that this library hasn't acquired full compatibility between SSD1306 and SH1106 drivers yet. As I researched online, I found out that there's a difference between the RAMs of SSD1306 and SH1106 drivers. This could be resolve simply by adding a macro (i.e. SH1106_DRIVER) in ssd1306_conf.h file. If user enables this macro, SSD1306_WIDTH should be 130 for OLEDs with 128-px width. In ssd1306_DrawPixel(..) function in ssd1306.c, the index for SSD1306_Buffer then becomes x + 2 + (y / 8) * SSD1306_WIDTH as follows:
```c
// Draw in the right color
    #if defined(SH1106_DRIVER)
    if(color == White) {
        SSD1306_Buffer[x + 2 + (y / 8) * SSD1306_WIDTH] |= 1 << (y % 8);
    } else { 
        SSD1306_Buffer[x + 2 + (y / 8) * SSD1306_WIDTH] &= ~(1 << (y % 8));
    }
    #else
    if(color == White) {
        SSD1306_Buffer[x + (y / 8) * SSD1306_WIDTH] |= 1 << (y % 8);
    } else { 
        SSD1306_Buffer[x + (y / 8) * SSD1306_WIDTH] &= ~(1 << (y % 8));
    }
    #endif
```

afiskon, 2021-01-13:
Thanks for the report. Could you please add links to the documents which describe these differences?

Dreamy-Z3r0, 2021-01-13:
The difference is described [here](https://forum.arduino.cc/index.php?topic=256374.0) by the user **olikraus**, reply \#4. I made some changes for the library you built with such information, and here's the result of correctly aligned display:

[image]

Here's what gets displayed by the library originally:

[image]

afiskon, 2021-01-15:
@Dreamy-Z3r0 I recognize this behavior. There use to be macro in the code that allowed to add/remove 2px shift required for SH1106. Unfortunately they were lost in one of pull requests. Thanks again for reporting this, I will fix it.

afiskon, 2021-01-15:
OK, I think I found it. If I remember correctly I used to change the definition of `SSD1306_WIDTH` for SH1106 https://github.com/afiskon/stm32-ssd1306/blob/master/ssd1306/ssd1306.h#L100 Maybe the README has to be more clear about how to use the library with SH1106.

afiskon, 2021-01-16:
I've added corresponding changes to ssd1306_conf.h

https://github.com/afiskon/stm32-ssd1306/blob/master/ssd1306/ssd1306_conf.h#L49

Please check if it works for you. The original patch you suggested, I'm afraid, causes out-of-memory reading, thus I can't make such changes in the code.

closed 2021-01-16 by afiskon