afiskon/stm32-ssd1306 #52: hi,

leesohaeng, 2021-12-11:
Hi
I am honored to use your code.

There is a problem with the ssd1306 test. 
I used your ssd3606 sample code and connected the ssd1306 using SPI. 
The code runs almost perfectly on SSD1306. 
However, there is always a vertical line on the right side of the screen. 

Any code fixes?
[image]

afiskon, 2021-12-11:
@leesohaeng hi! Try changing SSD1306_WIDTH as described [here](https://github.com/afiskon/stm32-ssd1306/blob/master/ssd1306/ssd1306_conf_template.h#L51) and re-compiling everything. Let me know if it helps.

leesohaeng, 2021-12-12:
> 
Hi
Solved. And I changed the starting position of x to 2 in some places.
Thank you very much for your help.
[image]

closed 2021-12-12 by afiskon