afiskon/stm32-ssd1306 #53: How to convert fonts to 16bits ?

cttneden, 2022-01-27:
May I ask what software did you use to convert the font to 16bits format?
code as shown below：
==============================================================
static const uint16_t Font7x10 [] = {
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,  // sp
0x1000, 0x1000, 0x1000, 0x1000, 0x1000, 0x1000, 0x0000, 0x1000, 0x0000, 0x0000,  // !
0x2800, 0x2800, 0x2800, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,  // "
0x2400, 0x2400, 0x7C00, 0x2400, 0x4800, 0x7C00, 0x4800, 0x4800, 0x0000, 0x0000,  // #
0x3800, 0x5400, 0x5000, 0x3800, 0x1400, 0x5400, 0x5400, 0x3800, 0x1000, 0x0000,  // $
================================================================

afiskon, 2022-01-27:
Sorry, this question duplicates https://github.com/afiskon/stm32-ssd1306/issues/3

closed 2022-01-27 by afiskon

cttneden, 2022-01-30:
Hi, afiskon:
Thanks for your reply.

________________________________
寄件者: Aleksander Alekseev ***@***.***>
寄件日期: 2022年1月27日 下午 03:56
收件者: afiskon/stm32-ssd1306 ***@***.***>
副本: cttneden ***@***.***>; Author ***@***.***>
主旨: Re: [afiskon/stm32-ssd1306] [Duplicate] How to convert fonts to 16bits ? (Issue #53)

Sorry, this question duplicates #3<https://apac01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fafiskon%2Fstm32-ssd1306%2Fissues%2F3&data=04%7C01%7C%7C9f537625d32a475d0e4a08d9e16a8abf%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637788669997628338%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=iRl5zHNYTQIZUHxlbNh2r%2FYwrnlYR7hYnrsDLXSjL3w%3D&reserved=0>