afiskon/stm32-ssd1306 #60: Possible endless cycle

HeatseekerXXX, 2022-12-27:
https://github.com/afiskon/stm32-ssd1306/blob/11c023452e2a9baa69069810fc11f32912daa3d5/ssd1306/ssd1306.c#L541

Hi. If someone tries to draw filled rectangle with coordinates starting with 0,0 (ex. FillRectangle(0,0,10,10,Black), he or she will be surprised. Program gets in an endless cycle.

```
for (uint8_t y = y2; y >= y1; y--) {
        for (uint8_t x = x2; x >= x1; x--) {
            ssd1306_DrawPixel(x, y, color);
        }
    }
```
Imagine x1=0 and x is constantly decreasing after each cycle. 
1. x=1, 1>=0 - true
2. x=0, 0>=0 - true
3. x=255, 255>=0 -true. "x" is an unsigned var.  
And so on.

By the way, I didn't search for problems in other similar functions. Maybe they should be reconsidered.

afiskon, 2022-12-27:
@HeatseekerXXX good catch, thanks for reporting it. Would you like to submit a fix?

HeatseekerXXX, 2022-12-29:
Made some changes in FillRectangle func that fixed issue. Also allowing to draw rect in any direction.

commit 04a656e6 (2022-12-30): Merge pull request #61 from HeatseekerXXX/master

FillRectangle fix. Closes #60
closed 2022-12-30 by afiskon