afiskon/stm32-ssd1306 #70: last 2 pixels at RHS of display remain on after reset

gharbarka, 2023-09-16:
I am new to this so most probably the error on my part. 

I have setup a 128x64 display with a blue pill running this software. I have watched the YouTube video several times to see if I have missed anything. In fact ssd1306_tests.c runs almost perfectly except for the error in caption. I also not that during the concentric rectangle test, the left hand side of the outermost rectangle is missing which leads me to think that the horizontal counter is getting shifted by a couple of pixels. I get the same result with a second display that I have. However these were bought together from the same supplier on Aliexpress so both could be faulty but I doubt it.

Any suggestions as to where to look for the mistake will be appreciated.

afiskon, 2023-09-16:
Thanks for reporting. This could be due to a couple of recent pull requests I accepted blindly. Could you please check if reverting the commit 5110f3e8 has any effect?

afiskon, 2023-09-16:
Also could you please show a photo of how it looks like? Some displays actually have "a shifted XY plane" in a manner of speaking - this is a well known situation and requires changing a few macro.

afiskon, 2023-09-16:
... particularly SSD1306_X_OFFSET - please see https://github.com/afiskon/stm32-ssd1306/blob/master/ssd1306/ssd1306_conf_template.h#L57

gharbarka, 2023-09-17:
Thank you for your speedy reply. Attached you will find 2 photos taken before the changes you proposed.

Reverting the commit did not solve the issue of the vertical border; but it did have an effect on the ssd1306_TestBorder() routine. Before the change the pixel traced a line along the top, then RHS, then Bottom right to left stopping in the middle of the line.  Deleting the 4 lines and inserting the 1 line, the pixel traced the complete right to left line.

What actually solved the issue was your second suggestion of setting an offset with SSD1306_X_OFFSET. In my case value of 2 was required. Interesting as well was the fact that after this change reverting the commit 5110f3e was not needed so I went back to the original for consistency.

Thanks again for your prompt help.

[Photos-001.zip](https://github.com/afiskon/stm32-ssd1306/files/12642461/Photos-001.zip)

afiskon, 2023-09-17:
🙏

closed 2023-09-17 by afiskon