afiskon/stm32-ssd1306 #76: Driver fails with 64x32 display

lamaatje, 2024-02-20:
The driver seems to fail with a 0.49" 64x32 Oled display. I verified i2c communication with a protocol analyzer. Certain commands seem to work like 0xAE (display off) 0xAF (display on). I also looked into ssd1306_Init() it seems not to follow the init sequence as specified in the datasheet on page 64 (Figure 2 : Software Initialization Flow Chart)  that make problem solving difficult. So I am wondering how 64x32 displays were verified.

afiskon, 2024-02-22:
No, the library was never tested with 0.49" 64x32 displays. Please make sure you are using proper SSD1306_WIDTH and SSD1306_HEIGHT values https://github.com/afiskon/stm32-ssd1306/blob/master/ssd1306/ssd1306_conf_template.h If you are certain there is a bug please feel free submitting a pull requests.

lamaatje, 2024-02-23:
Also got DMA working: 
[image]
@400kHz iic 4 pages (1 frame) takes around 7 ms.  Potential frame rate 140 Hz with zero CPU usage for frame transfer.

lamaatje, 2024-02-26:
as indicated ssd1306_Init() is:
1. not maintainable
2.  definitely not working with 64x32  and very likely also not working with other odd formats 

In case of issues with the proposed change finding an issue is relatively easy. With the current implementation it is like finding a needle in a haystack as I experienced myself. I took me more then 6 hours to get the 64x32 display working. ssd1306_Init() is a mess and completely diverging from the init in the datasheet. 

I2C DMA is nicely implemented with an optional compile switch and there is no dependency with a SPI implementation.

If the pull request is not acceptable for you I will maintain a separate fork or alternative implementation. Without DMA support the applicability of the library for "real" application is limited unless you want to burn CPU cycles with your 32-bit  microcontroller.

lamaatje, 2024-02-26:
Will maintain a fork.

closed 2024-02-26 by lamaatje