afiskon/stm32-ssd1306 #81: Custom-fonts Generate not working for me

tmargot, 2024-04-02:
Hi,
I have been trying to get the conversion working but keep getting the following error no matter what I try. I even tried copying the font file Teko-Bold.ttf font to the same folder as the program but no success. Any help would be appreciated. 

` C:\temp>.\generate.py --font Teko-Bold.ttf --size 20

File "C:\temp\generate.py", line 24
    match args.charset:
             ^
SyntaxError: invalid syntax`

` Directory of C:\temp

02/04/2024  12:52    <DIR>          .
02/04/2024  12:52    <DIR>          ..
01/04/2024  13:01             3,375 atlas.png
01/04/2024  13:01             1,512 convert.py
01/04/2024  13:01             4,705 generate.py
01/04/2024  13:01            37,422 hd44780-large.txt
01/04/2024  13:01             3,852 hd44780-small.txt
01/04/2024  13:01             2,638 README.md
01/04/2024  13:01               231 string.png
25/08/2023  18:51           154,868 Teko-Bold.ttf
01/04/2024  13:01             1,468 upscale.py
               9 File(s)        210,071 bytes
`

kossnikita, 2024-04-03:
Thanks for testing. This script requires Python 3.10+ for pattern matching.

tmargot, 2024-04-03:
Thank you @kossnikita for the reply, I updated my Python but then received the following error which I resolved by installing Pillow. On my way now, thanks again!

```
Traceback (most recent call last):
  File "C:\Other Programs\Python\font-converter\generate.py", line 1, in <module>
    from PIL import Image, ImageDraw, ImageFont
ModuleNotFoundError: No module named 'PIL'
```

PR #82 (2024-04-04): add font generator python version check
fix #81
closed 2024-04-03 by tmargot